MRX8024/chopper-resonance-tuner #1: Repo is corrupt

The-Original-Reth, 2024-03-05:
Hello, 
i get the following error on update manager in mainsail

Repo is dirty. Detected the following modifed files: ['.gitignore', 'Readme.md', 'chopper_plot.py', 'chopper_tune.cfg', 'gcode_shell_command.py', 'install.sh', 'wiki/EN.md', 'wiki/RU.md', 'wiki/chopper_tuning_guide_english.md', 'wiki/chopper_tuning_guide_russian.md', 'wiki/license.txt', 'wiki/manual_install_en.md', 'wiki/manual_install_ru.md', 'wiki/pictures/img_1.png', 'wiki/pictures/img_2.png', 'wiki/pictures/img_3.png', 'wiki/requirements.txt', 'wiki/wiki.md']

Along with the words "repo is corrupt"

I do a hard recovery and I get the following message

17:16:15 | Git Repo chopper-resonance-tuner: Attempting Repo Recovery...
-- | --
17:16:15 | Git Repo chopper-resonance-tuner: Starting Clone Recovery...
17:16:15 | Cloning into '/home/pi/.chopper-resonance-tuner_repo_backup'...
17:16:16 | Error Recovering chopper-resonance-tuner
17:16:16 | [Errno 13] Permission denied: 'gcode_shell_command.py'

How do I fix this. 
thank you

MRX8024, 2024-03-05:
Hello, thank you for issue feedback.
I fixed the "repo is corrupt" problems you described. Why you are experiencing problems with lack of access during recovery is difficult for me to answer. I will recommend you to force delete the repository: `sudo rm -rf ~/chopper-resonance-tuner/` and install it again from https://github.com/MRX8024/chopper-resonance-tuner/blob/main/wiki/chopper_tuning_guide_english.md.

Best regards
-Maxim

The-Original-Reth, 2024-03-05:
I get the following error when I try to use the install script

   command: /usr/bin/python3 /tmp/tmph2wtmr8s_in_process.py build_wheel /tmp/tmpp217v70t
       cwd: /tmp/pip-install-n0avch3j/pyarrow_a5186e642ff24df9a9cf1863e365c815
  Complete output (338 lines):
  <string>:34: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-cpython-39
  creating build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/cffi.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/benchmark.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/__init__.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/_generated_version.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/filesystem.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/substrait.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/json.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/acero.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/util.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/pandas_compat.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/ipc.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/types.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/compute.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/flight.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/dataset.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/fs.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/orc.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/hdfs.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/cuda.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/_compute_docstrings.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/conftest.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/jvm.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/csv.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  copying pyarrow/feather.py -> build/lib.linux-aarch64-cpython-39/pyarrow
  creating build/lib.linux-aarch64-cpython-39/pyarrow/parquet
  copying pyarrow/parquet/__init__.py -> build/lib.linux-aarch64-cpython-39/pyarrow/parquet
  copying pyarrow/parquet/core.py -> build/lib.linux-aarch64-cpython-39/pyarrow/parquet
  copying pyarrow/parquet/encryption.py -> build/lib.linux-aarch64-cpython-39/pyarrow/parquet
  creating build/lib.linux-aarch64-cpython-39/pyarrow/vendored
  copying pyarrow/vendored/__init__.py -> build/lib.linux-aarch64-cpython-39/pyarrow/vendored
  copying pyarrow/vendored/version.py -> build/lib.linux-aarch64-cpython-39/pyarrow/vendored
  copying pyarrow/vendored/docscrape.py -> build/lib.linux-aarch64-cpython-39/pyarrow/vendored
  creating build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_adhoc_memory_leak.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_cpp_internals.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/__init__.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_ipc.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_fs.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/pandas_examples.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_util.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_pandas.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_cuda.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/strategies.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_extension_type.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/util.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/pandas_threaded_import.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_substrait.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_csv.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/arrow_7980.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_deprecations.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_dlpack.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_io.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_cython.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_convert_builtin.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_dataset.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_scalars.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_array.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
  copying pyarrow/tests/test_memory.py -> build/lib.linux-aarch64-cpython-39/pyarrow/tests
[... 251 lines cut ...]
  -- Found Python3Alt: /usr/bin/python3
[... cut ...]
  See also "/tmp/pip-install-n0avch3j/pyarrow_a5186e642ff24df9a9cf1863e365c815/build/temp.linux-aarch64-cpython-39/CMakeFiles/CMakeOutput.log".
  See also "/tmp/pip-install-n0avch3j/pyarrow_a5186e642ff24df9a9cf1863e365c815/build/temp.linux-aarch64-cpython-39/CMakeFiles/CMakeError.log".
  error: command '/usr/bin/cmake' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for pyarrow
Failed to build pyarrow
ERROR: Could not build wheels for pyarrow which use PEP 517 and cannot be installed directly

MRX8024, 2024-03-05:
Pyarrow is an optional module, but recommended. Errors, fixes for installing it, and other packages for your specific OS image, you can look at various sites. Manually install the remaining packages from https://github.com/MRX8024/chopper-resonance-tuner/blob/main/wiki/requirements.txt using `sudo pip install`

-Maxim

The-Original-Reth, 2024-03-05:
Version: v0.12.0-113-g28f06a10-dirty
OS: Raspbian GNU/Linux 11 (bullseye)
Distro: MainsailOS 1.2.1 (bullseye) 

on a pi4

MRX8024, 2024-03-05:
I can recommend you to check the following packages - `pip install --upgrade pip setuptools` and
`sudo apt-get install build-essential libssl-dev`. In any case, this problem is not related to the original one, so I will close this issue.

closed 2024-03-05 by MRX8024