MRX8024/chopper-resonance-tuner #2: pyarrow not installing

The-Original-Reth, 2024-03-05:
I installed all the programs in requirement.txt except for pyarrow. 

I got the same error as before...

  ERROR: Failed building wheel for pyarrow
Failed to build pyarrow
ERROR: Could not build wheels for pyarrow which use PEP 517 and cannot be installed directly

If this is optional it might be best to remove the requirement, or give a way to install without it spitting out this error

MRX8024, 2024-03-05:
Hello again. One of these days I will install this OS image on my raspberry pi and will try to fix this error. This package is currently moved to the very bottom of the list of required packages (most recent action). So, even if it doesn't install, nothing bad will happen.

MRX8024, 2024-03-24:
Pyarrow was removed from the program in #4c64196

-Maxim

closed 2024-03-24 by MRX8024