MRX8024/chopper-resonance-tuner #6: If you have a linux32, do not spend time

EU4AX, 2024-03-22:
did I get that right? 
"sudo pip install pyarrow" ends in an error

https://arrow.apache.org/docs/python/install.html
Installing PyArrow
System Compatibility
PyArrow is regularly built and tested on Windows, macOS and various Linux distributions. We strongly recommend using a 64-bit system.

MRX8024, 2024-03-22:
Hello! I don't really understand the essence of your question. Do you mean that it makes no sense to try to install pyarrow on a 32bit OS? If it causes a wheel assembly error, then yes, it's probably not worth trying. 
P.S. I plan to rewrite some of the code to remove these modules.

-Maxim

EU4AX, 2024-03-22:
Hi! Thank you for your prompt response. Yes, I could not install Pyarrow on my 32-bit system. If this is not possible at this stage, perhaps this should be mentioned in *.md. 
However, the project looks very interesting. I look forward to a new installer.

MRX8024, 2024-03-24:
Hello, I meant the program code, not installer.  Actually, I rewrote it, and now it does not require pandas, which requires pyarrow.  If you have no questions left, please close the issue.

-Maxim

EU4AX, 2024-03-26:
now ok, tnx!

closed 2024-03-26 by EU4AX