MRX8024/chopper-resonance-tuner #16: Bad colors

anton-vinogradov, 2024-05-10:
Feel myself as colorblind %) 

[image]

MRX8024, 2024-05-13:
Hello. Xd, could you suggest a different color coding?

-Maxim

anton-vinogradov, 2024-05-13:
Seems, it should not be inscribed into the window, but be scrollable instead to avoid the colors overlay.

dougshell, 2024-05-15:
https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Fi2.wp.com%2Fteresabernardart.com%2Fwp-content%2Fuploads%2F2016%2F02%2FColorWheel-Tertiary.jpg&f=1&nofb=1&ipt=d19596a0ebab248bdbd75703e72f60ef03828565f3f85eb00730d5f04ee75c7f&ipo=images  use tertiary colors

anton-vinogradov, 2024-05-16:
It looks like problem not with colors, but with a attempt to show so many lines at once.
Let's just allow scrolling.

anton-vinogradov, 2024-05-16:
Colors are fine when zoomed.

MRX8024, 2024-05-16:
> regardless of the scroll problem, the colors are super desaturated. That is really the issue

I tested bright colors at some stage of the plotter, it was terrible. However, you can offer me some other colors, in html format, and I will test them.

> Let's just allow scrolling.

I'll try to see how to implement this.

dougshell, 2024-05-16:
@MRX8024 because the issue is mitigated by zooming out, possibly put a discalimer on the html graph. "For better color visibility, zoom out on web page until colors are more defined" This will then allow the user to select an appropriate range, and then zoom back in if needed