MRX8024/chopper-resonance-tuner #17: Timeout

Matszwe02, 2024-05-13:
I'm running my commands with RPI3. Chopper plot takes 20 minutes to complete, and in cfg file timeout is 999, causing it to break. Increase the timeout.

MRX8024, 2024-05-13:
Hello, wow, this is very long. With a full calibration with 1 iteration it should process the data in no more than 4-5 minutes.  Are you performing any actions on the host?, for example, home assistant, or a second printer, or something else..

Matszwe02, 2024-05-13:
No, printer was idle. When it finished, I saw the data completly flat, so my accel may be not seeing really anything, idk if that could be the cause

MRX8024, 2024-05-13:
Okay, increasing the timeout seems like the only option at the moment, but 20 minutes is creepy.  Could you provide graphs, tell us in more detail about your printer configuration, and the nearest information, and a klippy log.  Then maybe I can help you with flat graphs

Matszwe02, 2024-05-14:
Sorry, as I didn't get any reasonable results from my accelerometer, I removed everything. I have voron 0, so maybe the issue was too low max speed? (I had to decrease it because it ran out of space - 120mm)

MRX8024, 2024-05-14:
Did you mean your speed graph didn’t show any peaks? The bed area of 120mm should not affect this, because the resonance should be at lower speeds. In any case, you didn't tell us about your printer configuration, and you didn't include klippy.log: which motors you use, drivers, their settings, rotation distance of pulley and more etc.
I'll clarify just in case: It took 20 minutes to build a graph of registers, or speeds?  
Making a new speed graph takes a couple of minutes. 
Without all of the above, you understand for yourself, it will be impossible to figure it out. By the way, on my v0, when I ran this script for the first and last time with an accelerometer on my head, there was also no clear peak. However, I already had optimal registers there, but I never tried to attach the accelerometer to the motor.

-Maxim

Matszwe02, 2024-05-14:
ok, right, I didn't include any of the relevant info.

Basically standard V0, nothing changed with motion system.

I have an accelerometer on my toolhead, so the vibrations may be dampened. Also my registers could be already pretty fine, because I didn't see any significant peaks during the speed test.
[image]

After that second, 2 hour test, the registers graph command was timed out. It was executing one csv per 1-2 seconds. It was basically as flat as the first one. I use tmc2209, 32microsteps, interpolate on.

MRX8024, 2024-05-16:
Can I ask a question, why are you sending 32 microsteps if you are using interpolation? 
Well, you still haven't named the model of your stepper motors, but I can suggest that you attach an accelerometer to them. If this does not produce any results on the speed graph, further calibrations will be pointless.

-Maxim

Matszwe02, 2024-05-16:
motors are MS14HS5P4150-11

I think that I'll just keep on having my standard parameters, as attaching accelerometer to a motor is more tricky and time consuming. 

As per interpolation, I read in tmc_autotune that it can be turned on wihtout making that much of a difference in print quality, but I think I didn't set enough microsteps for that
https://github.com/andrewmcgr/klipper_tmc_autotune?tab=readme-ov-file#adjusting-existing-configuration