MRX8024/chopper-resonance-tuner #18: ''Type error'' error in chopper_plot.py

actualbigbobin, 2024-05-14:
When I run the chop_tune command it runs it for a little then I get this error:

Magnitude graphs generation...
Traceback (most recent call last):
File "/home/biqu/chopper-resonance-tuner/chopper_plot.py", line 159, in
main()
File "/home/biqu/chopper-resonance-tuner/chopper_plot.py", line 110, in main
data = np.array([[float(row["accel_x"]),
File "/home/biqu/chopper-resonance-tuner/chopper_plot.py", line 110, in
data = np.array([[float(row["accel_x"]),
TypeError: float() argument must be a string or a number, not 'NoneType'

Also the moment I run it I also get this error:

/home/biqu/chopper-resonance-tuner/chopper_plot.sh: line 4: /home/biqu/chopper-resonance-tuner/.venv/bin/activate: No such file or directory

MRX8024, 2024-05-15:
Hello, does the error occur in speeds, or registers test? 
As for the second, I corrected it, please update the repository and make install steps again.

-Maxim

actualbigbobin, 2024-05-15:
It occurs in registers test, speeds work fine

MRX8024, 2024-05-15:
Please show `df -h` output

MRX8024, 2024-05-15:
Yea, sorry, I keep forgetting to add this item to the wiki. You only have 494mb in /tmp when you need ~700, which stops writing content to files.. Use `sudo mount -o remount,size=1000M /tmp` to increase this size (works only until reboot).

actualbigbobin, 2024-05-15:
After I did that klipper UI gets stuck at Initializing, so I can't run the plotter command. I have a CB1 btw, if that makes a difference

MRX8024, 2024-05-15:
Yes, I saw the name of the device in the screenshot, in fact, so far I have only heard complaints from cb1.  I don’t know why btt made such a low (dynamic) space reserve /tmp on their image...

actualbigbobin, 2024-05-15:
Still the same issue after resizing tmp and updating the repository.

MRX8024, 2024-05-16:
Brr.. Scroll through the .csv in /tmp, do they all have weight? After entering the command to expand the directory, run `df -h` again, make sure that the size there has changed. There simply cannot be any other reason if the speed test works.

-Maxim

actualbigbobin, 2024-05-16:
228 of them are 0B

actualbigbobin, 2024-05-18:
Should I do it again?

MRX8024, 2024-05-19:
Hi, you still dont have enough space in /tmp. I have given you instructions to change and check this parameter. Everything is in your hands.

-Maxim