MRX8024/chopper-resonance-tuner #19: Влияние механической нагрузки двигателя и экземпляра драйвера на точность измерений. на точность измерений.

AlatusAles, 2024-05-19:
Здравствуйте! Спасибо, за хороший инструмент! Я прочитал [этот ответ](https://github.com/MRX8024/chopper-resonance-tuner/issues/9#issuecomment-2081391861), но возникли вопросы: Механическая нагрузка (ремни, балки, прочее) вносят свои резонансы, если закрепить двигатель на достаточно жёстком стенде с высокой собственной резонансной частотой это улучшит качество измерений? Вы говорите о "обязательной нагрузке" при измерениях, пожалуйста, в дух словах, опишите причину этого, к примеру: изменение  натяжения ремня может исказить результаты? Влияет ли конкретный экземпляр драйвера на измерения при прочих равных?

MRX8024, 2024-05-22:
Доброго, сложный вопрос. В теории улучшит результат, но по отношению к шаговому двигателю. Тут же мы скорее находим наименьшие колебания всей системы. Однако проводя эксперементы на совершенно разных станках, но на одной моделе моторов, результаты можно сказать сходились. Отсюда можно сделать вывод что механика не очень то на это и влияет.
Более сильное натяжение ремня заставит передавать большие колебания, но это все так же будет пропорционально для всех измеренных итераций. То, что мотор должен испытывать нагрузку - неоднозначный момент, который должен быть проверен физически. Однако дополнительная нагрузка на вал действительно может вызывать его раскачивания, запаздывания и просто более крупное обратное эдс. 
Экземпляр драйвера конечно влияет, но разница настолько незначительна, что акселерометр не сможет это засечь, как минимум при текущем способе измерений.

AlatusAles, 2024-05-22:
Приветствую! Спасибо, за ваше внимание! Протестировав в принтере, на одном из моторов, 5 экземпляров 2209, соглашусь о незначительности разницы "экземпляр-экземпляр" драйвера. Так получилось, что у меня около 30 килограмм :) различных шаговых двигателей которые я применяю в свих DIY проектах (3D принтеры в их числе) и в теории - использование вашего инструмента "на стенде" может помочь делать выборку лучших экземпляров двигателей (нецелесообразно устанавливать все тестируемые моторы в принтер...). На пути к стенду столкнулся с проблемой: так-как использовать одноплатник установленный в принтер желания нет, решил использовать ноутбук с Debian 12 (ноутбук использую для отладки всего, что делаю на Klipper), при установке вашим скриптом появляются  ошибки (примерно с 86-й строки скрипта) полагаю, у меня проблемы с ".venv" или "pip" или чем-то похожим, (стандартное построение графиков инпутшейпинга средствами Klipper работает хорошо). К сожалению, знаний о Linux у меня хватает только на копипасту команд... Заметил еще нюанс, мне не удалось полностью выполнить макрос с принтером (картезиан) у которого начало координат в центре стола. Расчет движения верный - (цифры примерные) "-138 -> 64: Длина 202", но после - "Out of range". Еще  раз, Спасибо!

MRX8024, 2024-05-23:
В таких случаях стоит копировать ошибки с терминала напрямую, либо весь лог установки. 
Исправил предполагаемые проблемы в установщике, пройдите повторно установку - `bash ~/chopper-resonance-tuner/install.sh`. 
Насчет картезиана аналогично - прикрепите klippy.log.

AlatusAles, 2024-05-23:
[Это](https://github.com/MRX8024/chopper-resonance-tuner/commit/3a627f4d990fc33815afa2466be87e55cd7b184c) работает! Спасибо!
 О кортезиан с нулем в центре стола - лог ниже.
[klippy.log](https://github.com/MRX8024/chopper-resonance-tuner/files/15416251/klippy.log)

AlatusAles, 2024-05-23:
И так, проведя всего 2 замера (не самого идеального двигателя) с нагрузкой и без нее я пришел к выводам: Форма графиков очень схожа -> для подбора двигателя тестированием на стенде, нагрузка не имеет большого значения, важен относительный результат; Нагрузка снижает амплитуду (в моем случае: примерно в 2 раза) и частоты резонансов  -> окончательное тестирование и корректировку параметров драйвера необходимо производить при полностью настроенной кинематике.
[image]
[image]

MRX8024, 2024-05-27:
> мне не удалось полностью выполнить макрос с принтером (картезиан) у которого начало координат в центре стола.

Доброго, исправлено в https://github.com/MRX8024/chopper-resonance-tuner/commit/bc4ae3e82d64c9e225cd71b7063cb025b3290554
Для обновления - `cd ~/chopper-resonance-tuner/ && git pull`, либо через веб морду.

AlatusAles, 2024-05-27:
Приветствую!
Благодарен!

"...использовать отсчет нуля с центра попахивает на мазохизм..." - Возможно :) Но: 
    Klipper этого не запрещает; 
    Привычка, т.к. большинство станков на работе ведут отсчет и в отрицательные числа координат, мне удобнее не менять начало координат в CAD/CAM/Slicer от задания к заданию; 
    Банально, удобнее при симметричных построениях и/или расчетах (например - [z_tilt] или [quad_gantry_level]...); 
    А вдруг не я один такой, а возможность уже реализована?!; 
    etc ...

Мой вопрос - решен.

Если, я могу помочь в развитии проекта (не кодом), сообщите.

closed 2024-05-27 by MRX8024