MRX8024/chopper-resonance-tuner #20: Adding test to move XY during the test

RocksaltX, 2024-05-20:
This is not an issue but a request. Is it possible to get the test to move 2 axis at one time as an option to only X or Y?
I would like to compare both movements at one time vs only 1 axis.

Great tool BTW! Thank you.

MRX8024, 2024-05-22:
Hello. It’s hard for me to imagine what you want to achieve and with what kinematics, describe it in more detail. Thanks.

-Maxim

RocksaltX, 2024-05-23:
Maxim,

I was looking to see the vibrations with both axis moving diagonal. Seems there is less noise when only one axis moves. I was thinking if I could see the diagonal move's vibrations, I could make changes to the drivers to see if it can be reduced.
This is a crossed gantry type printer.