MRX8024/chopper-resonance-tuner #22: Error after fresh install

ReaperXD1, 2024-05-29:
Hi there, I have just installed the chopper resonance tuner, however I continue to get this error when attempting to run the macro.
[image]

I don't know what is going on, I've even done the manual installation.

I am running a CB1 if that changes anything.

MRX8024, 2024-05-29:
Hi, pin a klippy.log

ReaperXD1, 2024-05-29:
Thanks for your quick reply. I've had to compress it as it was over githubs file size limit.
[klippy (3).zip](https://github.com/MRX8024/chopper-resonance-tuner/files/15487287/klippy.3.zip)

MRX8024, 2024-05-29:
Please update to the latest version, I have fixed some possible errors  there

ReaperXD1, 2024-05-30:
Still having this issue. Does it matter that I am using the BTT Klipper fork required for the new Eddy sensor?
[image]
[klippy.log](https://github.com/MRX8024/chopper-resonance-tuner/files/15492824/klippy.log)

MRX8024, 2024-05-30:
Try to install original klipper, which already support eddy. But it’s strange if btt broke something in their fork. And can you attach link to this fork please

ReaperXD1, 2024-05-30:
Original Klipper doesn't support the BTT Eddy unfortunately. 

https://github.com/bigtreetech/Eddy/tree/master - Eddy github
https://github.com/bigtreetech/klipper - BTT Klipper

MRX8024, 2024-05-30:
https://www.klipper3d.org/Config_Reference.html#probe_eddy_current

ReaperXD1, 2024-05-30:
I will try to get it working, give me a little :)

ReaperXD1, 2024-05-30:
Hi there, same story, on latest Klipper
[Uploading klippy (2).log…]()

MRX8024, 2024-05-30:
Since jinja2 cannot show the exact location of the error, I placed tags in the code. Please replace the macro, for example by replacing 'include' line
[chopper_tune_test.cfg.txt](https://github.com/MRX8024/chopper-resonance-tuner/files/15501745/chopper_tune_test.cfg.txt)

ReaperXD1, 2024-05-31:
Hi, attached is the new Klippy log.
[klippy (4).log](https://github.com/user-attachments/files/15507634/klippy.4.log)

MRX8024, 2024-05-31:
I think I found a mistake..
Try this - [chopper_tune_test.cfg.txt](https://github.com/user-attachments/files/15511216/chopper_tune_test.cfg.txt)

ReaperXD1, 2024-06-02:
Hi there I'll give it a shot now!

ReaperXD1, 2024-06-02:
It works! Thank you so much for your work! What was the issue?

MRX8024, 2024-06-02:
Script was supposed to get the sense_resistor value from config, which was not specified there. Now after c67b6c4 I take it from another place. Thanks for the feedback, if you have no questions left, I think we can close this issue.

-Maxim