MRX8024/chopper-resonance-tuner #24: Traceback Error After CHOPPER_TUNE MIN/MAX command

ReaperXD1, 2024-06-07:
Hi there, 

I have found another error, after waiting over 2 hours for the testing to finish twice, the terminal output an error;

MRX8024, 2024-06-07:
Please show `df -h` output, but by the name of the host I can already see what the problem is with memory limit

ReaperXD1, 2024-06-07:
[image]

ReaperXD1, 2024-06-07:
Ah okay, I'll give it another try maybe tomorrow.. It's a very loud process

MRX8024, 2024-06-07:
I didn't notice your last suggestion... if you haven't rebooted the host yet and the csvs are still in /tmp, you can .zip them and I'll generate a graph using the current data

ReaperXD1, 2024-06-07:
Luckily I found them in my /tmp and transferred them! Thank you very much!
https://drive.google.com/file/d/19LshR2UBQfhbOpDIkJGePYc3pNl5H7x5/view?usp=sharing

MRX8024, 2024-06-14:
Hello! Sorry for the long wait, Ive had a lot of work fall on me. I analyzed your graphs and also added code, now program would not throw an error, will work with the available data.
[issue24.zip](https://github.com/user-attachments/files/15843793/issue24.zip)

closed 2024-08-09 by ReaperXD1