MRX8024/chopper-resonance-tuner #26: How to best mount sensor

chrisheib, 2024-07-29:
Hi there, thanks for the work on this! 

I want to run this on my bedslinger, mainly my Y, as it's very noisy. 

Do you have advice on how to mount my sensor to the motor? Does the motor need to be mounted to the printer / under usual load?

Do you have any STLs for mounts?

MRX8024, 2024-07-29:
Hello, I just unscrewed one screw from the stepper, and then screwed in the accelerometer with it. I also did not remove the motor from the printer, and I have no other feedback, good luck!

-Maxim

chrisheib, 2024-07-30:
Ran the test, thanks again!

My results are a bit less impressive, but still really interesting:

[image]

[image]

chrisheib, 2024-07-30:
Unfortunately it the optimal settings don't seem to make any difference in noise: 

https://github.com/user-attachments/assets/7e2df85c-ab0c-451b-af43-e2496be777dc

MRX8024, 2024-07-30:
Hi, yes, these are very unusual results, the registers unfortunately have almost no effect on this motor. This is literally the lowest increase I've seen. Can you share klippy.log and your stepper motor model?

-Maxim

chrisheib, 2024-07-30:
This is a D42HS3418-24B22 on a heavily modified Geetech A20M with an SKR E3V3 mini.

Data for the stepper, its the first in the message: https://github.com/andrewmcgr/klipper_tmc_autotune/pull/162

Log: https://gist.github.com/chrisheib/e50d6b59616db68f027a5d15eabac2e1

chrisheib, 2024-07-30:
Ah, forgot i had the tmc_autotune still enabled for Y. Shouldnt make any difference in measuring tho, right? Will disable it and retry the noise test.

MRX8024, 2024-07-30:
In theory, no, autotune should not affect this, since it changes the same registers, but I have not read its code. Regarding the above case - have you tried turning off the stealthchop mode? Perhaps the motor works better in it, and will show a better reduction in vibrations from the registers.

-Maxim

chrisheib, 2024-07-31:
Disabling Stealthcop definitely had some effect and changed the sound profile a lot - overall a lot higher sound, with some static crispy sound during idle. I prefer the stealthchop sound.

Will interpolate have a big impact on sound?

chrisheib, 2024-07-31:
On X axis it looks way better.

[image]

I swapped the Y motor, should be worse on paper, but im not going for great speed anyway so it should be fine. Will run the optimization later.

chrisheib, 2024-08-01:
Im still a bit confused on the results. While X axis measurements look awesome, the real world results are... not there? 

Setting the parameters from the default 2-3-5-0 (around 3500 amplitude) to 0-8-1-8 (around 2200 amplitude) should reduce vibrations by over 30%, but there ist just no difference in motor behaviour o.0 Neither sound, nor sound quality, nor subjective vibrations change at all. 

Even when running a test gcode to change parameters on the fly there is no difference. 

Settings: 
```ini
[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
microsteps: 32
rotation_distance: 40
endstop_pin: ^!PC0
position_endstop: 0
position_min: 0
position_max: 245
homing_speed: 100

[tmc2209 stepper_x]
uart_pin: PC11
tx_pin: PC10
uart_address: 0
run_current: 1
stealthchop_threshold: 999999
interpolate: true
driver_TBL: 0 # default 2
driver_TOFF: 8 # default 3
driver_HSTRT: 1 # default 5
driver_HEND: 8 # default 0
```

Test gcode file:

```gcode
G28 X
SET_SKEW CLEAR=1

SET_TMC_FIELD STEPPER=stepper_x FIELD=TBL VALUE=0
SET_TMC_FIELD STEPPER=stepper_x FIELD=TOFF VALUE=8
SET_TMC_FIELD STEPPER=stepper_x FIELD=HSTRT VALUE=1
SET_TMC_FIELD STEPPER=stepper_x FIELD=HEND VALUE=8

G0 F15000 X200
G0 F15000 X50
G0 F15000 X200
G0 F15000 X50
G0 F15000 X200
G0 F15000 X50
G0 F15000 X200
G0 F15000 X50
G0 F15000 X200
G0 F15000 X50
G0 F15000 X200

SET_TMC_FIELD STEPPER=stepper_x FIELD=TBL VALUE=2
SET_TMC_FIELD STEPPER=stepper_x FIELD=TOFF VALUE=3
SET_TMC_FIELD STEPPER=stepper_x FIELD=HSTRT VALUE=5
SET_TMC_FIELD STEPPER=stepper_x FIELD=HEND VALUE=0

G4 P2000

G0 F15000 X200
G0 F15000 X50
G0 F15000 X200
G0 F15000 X50
G0 F15000 X200
G0 F15000 X50
G0 F15000 X200
G0 F15000 X50
G0 F15000 X200
G0 F15000 X50
G0 F15000 X200
```

chrisheib, 2024-08-01:
Again when disabling stealthchop the results are noticeable, but the optimal spreadcycle setting sounds worse than stealthchop. 

Maybe this tuning is only applicable when stealthchop is disabled.

MRX8024, 2024-08-02:
The above chart was carried out on stealthchop? In this case, the registers are applied correctly, and the accelerometer sees a strong reduction in vibrations. At F15k the results may be much less noticeable than at other ranges, both lower and higher, which may be why you don't see the difference.

-Maxim