MRX8024/chopper-resonance-tuner #29: Project is broken - I'd love to use it but I cannot

oernster, 2024-09-12:
I run the first command: CHOPPER_TUNE FIND_VIBRATIONS=1
and I get this: {"error": {"code": 404, "message": "Not Found", "traceback": "Traceback (most recent call last):\n\n  File \"/home/pi/moonraker-env/lib/python3.11/site-packages/tornado/web.py\", line 1790, in _execute\n    result = await result\n             ^^^^^^^^^^^^\n\n  File \"/home/pi/moonraker/moonraker/components/application.py\", line 795, in get\n    self.absolute_path = self.validate_absolute_path(\n                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n\n  File \"/home/pi/moonraker-env/lib/python3.11/site-packages/tornado/web.py\", line 2900, in validate_absolute_path\n    raise HTTPError(404)\n\ntornado.web.HTTPError: HTTP 404: Not Found\n"}}

I am running latest of everything as of 12/9/2024 and bookworm latest.

oernster, 2024-09-12:
I've just installed tornado using pip with --break-system-packages to see if that improves matters...

oernster, 2024-09-12:
no joy

MRX8024, 2024-09-12:
Hi! Please share the klipper logs manually or via https://coderus.openrepos.net/klipper_logs

oernster, 2024-09-12:
Please see here: https://coderus.openrepos.net/klipper_logs/366491c219f755498fb08a903d81ee1b

oernster, 2024-09-12:
or here: 
[klippylogs.zip](https://github.com/user-attachments/files/16974380/klippylogs.zip)

oernster, 2024-09-12:
I messed with the code in moonraker so I reverted it and this is the latest log.
[moonraker.log](https://github.com/user-attachments/files/16976035/moonraker.log)

oernster, 2024-09-12:
no longer applicable.

closed 2024-09-12 by oernster