MRX8024/chopper-resonance-tuner #30: KeyError in chopper_plot.py

eroboticdude, 2024-10-18:
Hello, I am trying this script on my Qidi q1 pro, and I had some issues, I got past most of them by overriding the default highest version of python to 3.9 instead of 3.7, which allowed me to get the required package versions.
After fixing the package errors, I now get this error:

```
/ Running Command {chop_tune}...:
09:19:58 
// Magnitude graphs generation...
// Traceback (most recent call last):
// File "/home/mks/chopper-resonance-tuner/chopper_plot.py", line 127, in
09:19:58 
// main()
// File "/home/mks/chopper-resonance-tuner/chopper_plot.py", line 71, in main
// static_data = calc_static_magnitude(file)
// File "/home/mks/chopper-resonance-tuner/chopper_plot.py", line 45, in calc_static_magnitude
09:19:59 
// data = np.array([
// File "/home/mks/chopper-resonance-tuner/chopper_plot.py", line 46, in
// [float(row["accel_x"]),
// KeyError: 'accel_x'
09:19:59 
// Command {chop_tune} finished
```
I think its possible the path for my tmp file is strange, as the q1 pro is based on an older version of klipper. I also had to edit the macro to not specify 'adxl345' as the chip name, because apparently in this version of klipper that is an invalid name. The script works for the most part, just the graph generation is not working properly.

eroboticdude, 2024-10-18:
I did try the tmp expansion mentioned in other issues, but all the csv files were intact so I believe that the size of the tmp directory isn't an issue

eroboticdude, 2024-10-18:
[adxl345-stand_still.csv](https://github.com/user-attachments/files/17437611/adxl345-stand_still.csv)

Is it possible that my old version of klipper formats the csv file differently? 
If I modify the calc_magnitude function to read columns instead of rows will it work?

eroboticdude, 2024-10-18:
after more looking, it seems like the csv function doesn't expect the first row, which has the drop count and other info
```
<html>
<body>
<!--StartFragment--><google-sheets-html-origin><style type="text/css"><!--td {border: 1px solid #cccccc;}br {mso-data-placement:same-cell;}--></style>
##drops=8 | overflows=0 | time_per_sample=0.000316789 | start_range=0.000007 | end_range=0.000007
-- | -- | -- | -- | --
#time | accel_x | accel_y | accel_z |  
109.844789 | 841.41057 | 0 | 9714.46749 |  
109.845105 | -152.98374 | 152.98374 | 0 |  
109.845422 | -152.98374 | 229.47561 | -76.49187 |  
109.845739 | 0 | 0 | 0 |  
109.846056 | 152.98374 | 0 | -76.49187 |  
109.846373 | 152.98374 | 0 | -229.47561 |  
109.846689 | 0 | 0 | 0 |  

<!--EndFragment-->
</body>
</html>
```
What is the best way around this? Mainline klipper is currently not easy to get on this printer, so I would like to do what I can with the current setup.

MRX8024, 2024-10-19:
Hi, switch branch: cd ~/chopper-resonance-tuner && git checkout qidi && git pull --force, try to check the work.
-Maxim

closed 2024-10-23 by eroboticdude