MRX8024/chopper-resonance-tuner #31: Qidi branch issues with install

eroboticdude, 2024-10-22:
Qidi printers are stuck with  python 3.7
Its possible to force an update to 3.9+, but it can cause issues and harm the function of the Qidi firmware. This is the error I get with install:

  ```
Cache entry deserialization failed, entry ignored
  Could not find a version that satisfies the requirement numpy>=1.26.3 (from -r                      /home/mks/chopper-resonance-tuner/wiki/requirements.txt (line 1)) (from version                     s: 1.3.0, 1.4.1, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.7.0, 1.7.1, 1.7.2, 1.8.0,                      1.8.1, 1.8.2, 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.10.0.post2, 1.10.1, 1.10.2, 1.10.4,                      1.11.0, 1.11.1, 1.11.2, 1.11.3, 1.12.0, 1.12.1, 1.13.0, 1.13.1, 1.13.3, 1.14.0,                      1.14.1, 1.14.2, 1.14.3, 1.14.4, 1.14.5, 1.14.6, 1.15.0, 1.15.1, 1.15.2, 1.15.3,                      1.15.4, 1.16.0, 1.16.1, 1.16.2, 1.16.3, 1.16.4, 1.16.5, 1.16.6, 1.17.0, 1.17.1,                      1.17.2, 1.17.3, 1.17.4, 1.17.5, 1.18.0, 1.18.1, 1.18.2, 1.18.3, 1.18.4, 1.18.5,                      1.19.0, 1.19.1, 1.19.2, 1.19.3, 1.19.4, 1.19.5, 1.20.0, 1.20.1, 1.20.2, 1.20.3,                      1.21.0, 1.21.1, 1.21.2, 1.21.3, 1.21.4, 1.21.5, 1.21.6)
No matching distribution found for numpy>=1.26.3 (from -r /home/mks/chopper-reso                     nance-tuner/wiki/requirements.txt (line 1))

```
The python libraries used require higher python levels than 3.7.
Is it possible to use different libraries that do not require update to higher python versions?
This would be a super cool script to remove the common VFAs on qidi printers.

eroboticdude, 2024-10-22:
the qidi branch is used currently, but there is only one small error I was having fixed, after I force python 3.9. This caused my printer to be bricked, and I had to do a full firmware flash to repair it.

eroboticdude, 2024-10-22:
numpy 1.21 is the highest python 3.7 supports,
matplotlib 3.7.x is also probably needed

MRX8024, 2024-10-22:
> and I had to do a full firmware flash to repair it.

Do you mean the host firmware?

eroboticdude, 2024-10-22:
> > and I had to do a full firmware flash to repair it.
> 
> Do you mean the host firmware?

Yeah, the Qidi Q1 uses a Mks Skipr clone. I must have deleted something important while i was installing python 3.9. No worries though, it was simple to repair

MRX8024, 2024-10-22:
Well, I lowered the requirements to 3.7, if you have the opportunity to test it, it would not be bad

eroboticdude, 2024-10-22:
> Well, I lowered the requirements to 3.7, if you have the opportunity to test it, it would not be bad

Thank you! I will test it and report any errors here

eroboticdude, 2024-10-22:
running the install script works, the only error to show up is this:
`Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-tqp7p8tz/matplotlib/`

hopefully that isn't a fatal error. We'll see in a moment

eroboticdude, 2024-10-22:
Unfortunately there seems to be more to this error:
```
 Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-tqp7p8tz/matplotlib/setup.py", line 349, in <module>
        "sdist": Sdist,
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/setuptools/__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/setuptools/dist.py", line 724, in fetch_build_eggs
        replace_conflicting=True,
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 782, in resolve
        replace_conflicting=replace_conflicting
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1065, in best_match
        return self.obtain(req, installer)
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1077, in obtain
        return installer(requirement)
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/setuptools/dist.py", line 791, in fetch_build_egg
        return cmd.easy_install(req)
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/setuptools/command/easy_install.py", line 704, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/setuptools/command/easy_install.py", line 730, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/home/mks/chopper-resonance-tuner/.venv/lib/python3.7/site-packages/setuptools/command/easy_install.py", line 901, in install_eggs
        os.path.abspath(dist_filename)
    distutils.errors.DistutilsError: Couldn't find a setup script in /tmp/easy_install-tf4w1xe6/numpy-2.1.2.tar.gz

    Edit mplsetup.cfg to change the build options; suppress output with --quiet.

    BUILDING MATPLOTLIB
          python: yes [3.7.3 (default, Mar 23 2024, 16:12:05)  [GCC 8.3.0]]
        platform: yes [linux]
           tests: no  [skipping due to configuration]
          macosx: no  [Mac OS-X only]

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-tqp7p8tz/matplotlib/
```

eroboticdude, 2024-10-22:
I forgot to mention a change I made to the original files on my last test!
QIDI Klipper uses a different location for its config files:
It places the config files in a directory called "klipper_config" 
I will do the procedures for manual install to continue testing, but modifying the install script would be cool

MRX8024, 2024-10-22:
> Unfortunately there seems to be more to this error:

Try rebooting the host, it looks like the installation package is broken

> called "klipper_config"

To ~/klipper_config ?

eroboticdude, 2024-10-22:
Yes, I will share the edits I made to the config and other files. First test was successful after manual installing of all the packages