MRX8024/chopper-resonance-tuner #32: Creality K1C Проблемы с установкой

Mannokia, 2024-10-24:
Добрый день. Я так понимаю проблема в том что на данном принтере стоит не стандартный Klipper и пути к папкам отличаются. А потому невозможно установить. 
Может ли кто нибудь помочь адаптировать скрипт для Creality K1C ?

MRX8024, 2024-10-24:
Доброго вечера, я понятия не имею какое оборудование и софт установлены на этом принтере, и с какими проблемами вы столкнулись при установке. Если будет предоставлена какая либо информация, можно будет попытаться разобраться в проблеме. Так же в инструкции описан ручной способ установки, если вам хватает навыков это реализовать.

Mannokia, 2024-10-27:
> Доброго вечера, я понятия не имею какое оборудование и софт установлены на этом принтере, и с какими проблемами вы столкнулись при установке. Если будет предоставлена какая либо информация, можно [...]

По нитке пытаюсь разобраться. Мне удалось в ручную установить по инструкции, меняя пути на те которые есть у меня.
Но я остановился  на 5 пункте с установкой пакетов.

Моя система как я смог выяснить `Linux K1C-D79C 4.4.94 #371 SMP PREEMPT Tue Jun 18 20:12:12 CST 2024 mips GNU/Lin                                          ux` И я полагаю что `sudo apt-get install` для меня не подходит. Потому как  `sudo apt-get install libatlas-base-dev libopenblas-dev` возвращает  `/usr/bin/sudo: exec: line 2: apt-get: not found`
Помогите пожалуйста, как мне установить недостающие 2 пакета?