MRX8024/chopper-resonance-tuner #34: CoreXY Diagonals

Anima1000000, 2024-12-06:
Would it be possible to test one motor at a time on CoreXY by just testing the diagonals?

MRX8024, 2024-12-06:
Hi, this is not possible at the moment. You can only change the kinematics type to cartesian, manually set the head coordinates, and run the test in such a “dangerous” way.

eoyilmaz, 2026-01-17:
I'm not sure if the original developers going to like the idea or not, but I completely re-written the system as a Klipper extension, introduced a lot of improvements like:

- `/tmp` is not used anymore, all the files are moved to `~/printer_data/config/adxl_results/chopper_magnitude/tmp` as soon as they are created and Klipper finishes dumping the data, so there is no chance of crashing your printer, just because no memory is left. So, the memory requirement is gone now.
- As explained above, the system waits for the resonance data to be dumped adaptively, so the fixed length of 0.5 seconds of wait plus moving back to the initial position has been removed during the initial resonance measurement phase, including the next change this speeds the measurements up.
- Introduced a coordinate generator, which helps measuring the resonancces/vibrations in both direction, essentially reducing the total run time, and also considering the other direction on selecting the best register values. This is not a Python Generator yet, I'll update it to be one soon.
- Introduced the `ACCELEROMETER_MEASURE` as a context manager, which elegantly starts and stops the measurements.
- I think the code is easier to read, understand and maintain as all the functionalities are now residing in their own functions/methods, and I'm keep imrpoving it.
- The new code is easier to indtroce the optimizer and I'll do it soon, this is going to be the ultimate enhancement to this system, as I'm contemplating the optimization will take a couple of minutes instead of hours, which will also allow more iterations to be made to improve the measurement quality.

So, yes, exciting times ahead...

eoyilmaz, 2026-01-19:
Hey @MRX8024 thanks for the comment, I created the PR and explained what could be done to keep the tool inline with your intentions of usage 👍