MRX8024/chopper-resonance-tuner #36: Timer too Close

Avatarsia, 2024-12-25:
Hi, 
i use a voron with a recore a6 mainboard (https://wiki.iagent.no/wiki/Recore_A6)
this usses a ar100 to controll the stepper drivers. 
when i run 
CHOPPER_TUNE MIN_SPEED=203 MAX_SPEED=203
it procede until 
CHOPCONF: 340204d1 toff=1 hstrt=5 hend=9 vsense=1 mres=4(16usteps) intpol=1 dedge=1
then the MCU 'ar100' shutdown: Timer too close message arrive. 

the ram is 247/900 mb
the swap is around 22.3 mb / 460
and the cpu idles around 20% 

is there a way to prevent that from happening ?

MRX8024, 2024-12-25:
Hi, does this happen every time?
Cool control board, never heard of such.
Attach klippy.log, maybe I'll find something in it.

Avatarsia, 2024-12-25:
okay small update, right before i wanted to prepare the log files it seems my board died. 
i changed it now to a new one and it seems to continue much longer