MRX8024/chopper-resonance-tuner #37: More Setup Information

DrWho-Capaldi, 2024-12-30:
I was able to get all the way down to adding the configuration into my printer.cfg file but it is not clear how these should be labeled:

[tmc**** stepper_*]
cs_pin: PC4

I get the tmc2209 or tmc2240 but what stepper is this assigned to? I'm running a Voron 2.4R2 Pro + with the BTT Manta M8P+CB1 board and the EBB SB2209 CAN (RP2040) along with the Cartographer 3D.  Can you post a full printer.cfg example instead of a partial?  I also have the TPFD in both my 2240 and the Cartographer (the Cartographer is the one currently enabled).

Thanks in advance.

MRX8024, 2025-01-04:
Hello, I didn't really understand the question. You apply registers to all drivers that are involved in toolhead movement during calibration. That is, in the case of voron and corexy, you apply registers to X and Y drivers at the same time.
Let me remind you that this utility is an experimental assistant tool to give an understanding of the influence of registers on the stepper motor and driver. And I hope that you are not using different drivers for the X Y axes :/ .

-Maxim

DrWho-Capaldi, 2025-01-04:
> Hello, I didn't really understand the question. You apply registers to all drivers that are involved in toolhead movement during calibration. That is, in the case of voron and corexy, you apply [...]

I'm going to be honest with you, that reply went clean over my head.  I'm not a programmer.  I'm trying to learn and configure my printer to print better and saw the post so I wanted to give it a try.  The issue is while I can add things to my printer configuration file and do have some understanding of how it works, I didn't have a clear picture of the exact configuration that was needed.  To be honest, since Voron and Klipper configurations are generic, I have pieced together a working configuration file but I know this printer can be so much better.  Your project seemed to be one that would help, I just don't understand how to implement it in my configuration.

MRX8024, 2025-01-04:
You can view how to configurate and study the params that the drivers support in the [klipper config reference](https://www.klipper3d.org/Config_Reference.html#tmc2209).
In this case, the calibrator helps to calibrate the following registers:
```
#driver_TBL: 2
#driver_TOFF: 3
#driver_HEND: 0
#driver_HSTRT: 5
```
 with default values.