MRX8024/chopper-resonance-tuner #38: IndexError: list index out of range

lusl13, 2025-01-04:
hello. have this problem:

10:51
main()
File "/home/biqu/chopper-resonance-tuner/chopper_plot.py", line 110, in main
speed1 = params[1][0][0].split('_')[6].split('=')[1]
IndexError: list index out of range

10:51
Magnitude graphs generation...
Traceback (most recent call last):
File "/home/biqu/chopper-resonance-tuner/chopper_plot.py", line 127, in 
10:51

Running Command {chop_tune}...:

graph was empty
[image]

[klippy (10).log](https://github.com/user-attachments/files/18306387/klippy.10.log)
[moonraker (2).log](https://github.com/user-attachments/files/18306388/moonraker.2.log)

MRX8024, 2025-01-04:
Hi, hmm, is this during the first test with speeds? 
Run it again and show what the `ls -la /tmp` command in linux terminal will produce after plot fail; fully.

lusl13, 2025-01-04:
no. in the second test. the first one passed fine

MRX8024, 2025-01-04:
If you didn't start a new one after the test, or didn't reboot the host, you can still run this command. Or repeat it again, if unsuccessful, download all files from `/tmp` and provide them in a zip archive.

lusl13, 2025-01-05:
after reboot everything works fine. chopper tuned. thank you.

closed 2025-01-05 by MRX8024