MRX8024/chopper-resonance-tuner #40: Warning!!! Empty data cells detected (75), make sure you dont run out of memory

DuckersMcQuack, 2025-03-22:
What does this mean? Dying sd card? As it can't be the ram, as i use a 8GB pi with 500MB ram used.

Magnitude graphs generation...
Access to interactive plot at: /home/pi/printer_data/config/adxl_results/chopper_magnitude/interactive_plot_adxl345_tmc2209_0.11_20250322_162949.html
Warning!!! Empty data cells detected (75), make sure you dont run out of memory

MRX8024, 2025-03-22:
This means that the /tmp folder in which the measurements are placed has run out of space. Klipper doesn't take this into account at all and creates empty files, so the only thing the script can do is warn the user about the problem after collecting the all data. Even if you have 8GB of RAM, depending on your OS, the /tmp folder has a certain limit, in general can be mounted even on non-ram, and it seems that in your case it is very small or occupied by some files. By `df -h` command you can check it.

DuckersMcQuack, 2025-03-22:
pi@Blender:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            3.6G     0  3.6G   0% /dev
tmpfs           781M  1.3M  780M   1% /run
/dev/mmcblk0p2  118G   12G  101G  11% /
tmpfs           3.9G  212K  3.9G   1% /dev/shm
tmpfs           5.0M   16K  5.0M   1% /run/lock
/dev/mmcblk0p1  510M   67M  444M  14% /boot/firmware
tmpfs           781M   48K  781M   1% /run/user/1000

So it just has 781MB it seems. Is there a way to increase this?

MRX8024, 2025-03-23:
781mb is enough, the problem is that your tmpfs is mounted anywhere but `/tmp`, where klipper saves its data:
```
filename = "/tmp/%s-%s.csv"
```
You don't have `/tmp` at all ? :
```
ls -ld /tmp
```
Create it:
```
sudo mkdir -p /tmp
sudo chmod 1777 /tmp
```
And if you want also move `/tmp` to `tmpfs`, see how to work with `/etc/fstab`

DuckersMcQuack, 2025-03-23:
pi@Blender:~ $ filename = "/tmp/%s-%s.csv"
-bash: filename: command not found

Made a new /tmp/ directory as it didn't exist as per your commands
Tried to rerun this command, but resulted in the same error

 `RUN_SHELL_COMMAND CMD=chop_tune PARAMS='iterations=1 driver=2209 sense_resistor=0.110000'`

[image]

MRX8024, 2025-03-23:
You must full retake the measurements. If you after also get an error, enter the command:
```
ls -lh /tmp
```
and show its output.

DuckersMcQuack, 2025-03-23:
The repeated csv's extends above that list a bit as well.

[image]

MRX8024, 2025-03-23:
Static measurement has a good weight of 728k bytes, but all subsequent measurements except partially the first have its 30, most likely without data at all. You can open it as it is a regular excel file. I can assume that your wire comes off the adxl during movement, perhaps some interference prevents the data from reaching, there can be many reasons.
But in general, this is not a problem of my script, but of the OS, klipper or hardware.

DuckersMcQuack, 2025-03-23:
No wires comes off. But i'm gonna try reinstalling the script and dependencies, if that won't work, i'm just gonna stop attempting and instead try different motors.

DuckersMcQuack, 2025-03-23:
Reason found! I shall report in 1 month's time when i've received the new adxl :P

[image]