MRX8024/chopper-resonance-tuner #41: Where is best place to ask for support?

yonkiman, 2025-04-05:
I don't want to spam github issue tracking with support requests, but it's not clear where else to go.  

It seems to have installed correctly on my Voron 2.4, but when I run CHOPPER_TUNE FIND_VIBRATIONS=1, the Y axis moves to mid way (about 150), then X moves to about 150, then X starts increasing, it seems to do about 3 measurements, then it tries to send X too far and ends with a `!! Move out of range: 359.240 164.500 5.000 [0.000]`.  Likewise when I try to test the Z axis, it does a few measurements then tries to go beyond my max Z.

I don't see any parameters in chopper_tune.cfg to control limits (except maybe `variable_inset: 10`, but I'm not sure what that even does)

One possible reason might be that I do not have all my config info in printer.cfg - I have it spread over multiple .cfg files.  But when I put [stepper_x]  (and y and z) back into printer.cfg and reboot klipper, the macro still tries to move past the 300mm limit.  

Is there more information on this anywhere?

MRX8024, 2025-04-05:
Contacts are signed at the end [EN.md](https://github.com/MRX8024/chopper-resonance-tuner/blob/main/wiki/EN.md). Github issue is not a bad place to discuss problems, but if you want you can contact me using the contacts described.
First of all, I'll need `klippy.log` anyway. 

-Maxim

yonkiman, 2025-04-05:
OK, thanks.  Looking at chopper_tune.cfg, I found the **Protect exceeding axis limits & calculate travel distance** section.  It looks like maxAX (at least) is not correct.  I'd try to debug myself, but my head explodes every time I try to understand variables in the klipper macro language.

This is the log file for starting klipper, running **CHOPPER_TUNE FIND_VIBRATIONS=1 axis=z**, which ended with **!! Move out of range: 160.000 10.000 263.233 [0.000]**.

[klippy.log](https://github.com/user-attachments/files/19617685/klippy.log)

MRX8024, 2025-04-06:
I took a quick look, well, it does look weird. I would guess that jinja2 in kalico works differently and at some point in the program undefined behavior occurs. Show what the terminal shows when running the test, M118 are not logged. I'll try to watch it, but I'm a bit busy in the next few days.

yonkiman, 2025-04-06:
Terminal output below.  Seems to read at least some of my config fine (drivers, accelerometer, run current, and all 4 Z motors are working).   

One question: when it's about to test the Z axis, it moves from (0,0) to (~160,0) (it just "centers" the X axis, doesn't move the Y).  Sort of wonder if it's trying to move to (150,150) but doesn't know it's a CoreXY. 

Thanks for any help.

17:26:31  $ CHOPPER_TUNE FIND_VIBRATIONS=1
17:26:31  // Selected tmc2209 for stepper_z
17:26:31  // Selected 4wd configuration
17:26:31  // Selected adxl345 for accelerometer
17:26:31  // Set default run_current: 800 mA to run_current_min
17:26:31  // Set default run_current: 800 mA to run_current_max
17:26:31  // Acceleration & deceleration zone = 1.142857 mm
17:26:31  // Auto calculated min required travel distance = 26.142857 mm
17:26:31  // minAX = 10.000000 mm
17:26:31  // Final max travel distance = 26.14 mm, position min = 10.00, traveling: 10.00 --> 36.14
17:26:31  // Start find vibration mode, speed: 5.00  --> 20.00 mm/s with 0.20 step current=800 TBL=2 TOFF=3 HSTRT=5 HEND=0
17:26:31  // Running Command {chop_tune}...:
17:26:31  // Command {chop_tune} finished
17:26:31  // accelerometer measurements started
17:26:41  // Writing raw accelerometer data to /tmp/adxl345-stand_still.csv file
17:26:41  echo: Setting curr from -1 to 800 on stepper_z
17:26:41  // Run Current: 0.80A Hold Current: 0.80A Home Current: 0.20A
17:26:41  echo: Setting curr from -1 to 800 on stepper_z1
17:26:41  // Run Current: 0.80A Hold Current: 0.80A Home Current: 0.20A
17:26:41  echo: Setting curr from -1 to 800 on stepper_z2
17:26:41  // Run Current: 0.80A Hold Current: 0.80A Home Current: 0.20A
17:26:41  echo: Setting curr from -1 to 800 on stepper_z3
17:26:41  // Run Current: 0.80A Hold Current: 0.80A Home Current: 0.20A
17:26:41  echo: Setting tbl from -1 to 2 on stepper_z
17:26:41  echo: Setting tbl from -1 to 2 on stepper_z1
17:26:42  echo: Setting tbl from -1 to 2 on stepper_z2
17:26:42  echo: Setting tbl from -1 to 2 on stepper_z3
17:26:42  echo: Setting toff from -1 to 3 on stepper_z
17:26:42  echo: Setting toff from -1 to 3 on stepper_z1
17:26:43  echo: Setting toff from -1 to 3 on stepper_z2
17:26:43  echo: Setting toff from -1 to 3 on stepper_z3
17:26:43  echo: Setting hend from -1 to 0 on stepper_z
17:26:43  echo: Setting hend from -1 to 0 on stepper_z1
17:26:44  echo: Setting hend from -1 to 0 on stepper_z2
17:26:44  echo: Setting hend from -1 to 0 on stepper_z3
17:26:44  echo: Setting hstrt from -1 to 5 on stepper_z
17:26:44  echo: Setting hstrt from -1 to 5 on stepper_z1
17:26:45  echo: Setting hstrt from -1 to 5 on stepper_z2
17:26:45  echo: Setting hstrt from -1 to 5 on stepper_z3
17:26:45  // CHOPCONF:   33030053 toff=3 hstrt=5 tbl=2 vsense=1 mres=3(32usteps) intpol=1 dedge=1
17:26:45  echo: Speed 5.0 mm/s on 6.54 mm
17:26:45  // accelerometer measurements started
17:26:49  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_500_41666_1__.csv file
17:26:50  echo: Speed 5.2 mm/s on 6.8 mm
17:26:50  // accelerometer measurements started
17:26:54  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_520_41666_1__.csv file
17:26:55  echo: Speed 5.4 mm/s on 7.06 mm
17:26:55  // accelerometer measurements started
17:26:55  !! Move out of range: 159.994 10.000 85.391 [0.000]
17:26:55  !! Move out of range: 159.994 10.000 85.391 [0.000]
17:26:55  !! Move out of range: 159.994 10.000 85.391 [0.000]

MRX8024, 2025-04-06:
> Sort of wonder if it's trying to move to (150,150) but doesn't know it's a CoreXY.

No, it's normal. The program was originally written only for corexy and only for the x-axis, then some diversity appeared, which was implemented like this.

> 17:26:55 echo: Speed 5.4 mm/s on 7.06 mm

Well, I thought you were seeing something different here, but no. 

> !! Move out of range: 159.994 10.000 85.391

In this case it reported `85.391` position z, did you reduce the limits in the configuration?
Otherwise I don't understand what's going on here at all, I would suggest you update/reinstall kalico, but in general I don't guarantee the program will work on kalico. You could easily install a klipper next to it and check on it, in which case the bug would definitely be in kalico. 

Sorry, I have nothing to add, everything looks legal.

yonkiman, 2025-04-06:
Thanks Maksim.  Yes, I'd set the max Z to 80mm so the test would end (fail) quicker.

I'm slowly getting up to speed on the Klipper macro language so I hope to be able to debug and figure it out soon.  I'll let you know if I do.

MRX8024, 2025-04-06:
I can offer you to add a something like this:
```
M118 Moving on {axes[0]} axis on {minAX + travel_distance} mm distance
```
after this [line](https://github.com/MRX8024/chopper-resonance-tuner/blob/1f98212ca9dbfdf15d516115dd4c26e97b914a8d/chopper_tune.cfg#L387), and show terminal output.

yonkiman, 2025-04-06:
Now I'm wondering if I don't understand how this is supposed to work.  I assumed that you are trying so many combinations of settings that (for Z axis, say), you must go up and down many times to capture all the combinations.  But the code makes it look like everything will happen during one "trip" along the axes:

        {% if minAX + auto_travel_distance > maxAX %}
            {action_raise_error('WARNING!!! Required travel distance on axis (%s)'
             ' (%.2f mm) is longer than kinematics allows, please lower'
              ' speed or increase acceleration' % (axes[0], auto_travel_distance))}

So maybe I just have it configured wrong?  On my machine it moves slow, then fast maybe 10 times (with Z set to 260) before I get the out of bounds error.  Is there a video of anyone performing this test?

BTW, thanks for creating this - it's a great idea and I'm hoping it will help reduce my Z axis noise so my Revo PZ probe can move faster and be more sensitive.

(And I'll try the M118 line in a bit.)

yonkiman, 2025-04-06:
Here's my console output with the M118 line and Zmax set to 260

 16:37:06  $ CHOPPER_TUNE FIND_VIBRATIONS=1
16:37:06  // driver = 2209
16:37:06  // Selected tmc2209 for stepper_z
16:37:06  // min speed = 0.110000 mm
16:37:06  // Selected 4wd configuration
16:37:06  // Selected adxl345 for accelerometer
16:37:06  // Set default run_current: 800 mA to run_current_min
16:37:06  // Set default run_current: 800 mA to run_current_max
16:37:06  // Acceleration & deceleration zone = 1.142857 mm
16:37:06  // Auto calculated min required travel distance = 26.142857 mm
16:37:06  // Final max travel distance = 26.14 mm, position min = 10.00, traveling: 10.00 --> 36.14
16:37:06  // Start find vibration mode, speed: 5.00  --> 20.00 mm/s with 0.20 step current=800 TBL=2 TOFF=3 HSTRT=5 HEND=0
16:37:06  // Running Command {chop_tune}...:
16:37:07  // Command {chop_tune} finished
16:37:07  // accelerometer measurements started
16:37:17  // Writing raw accelerometer data to /tmp/adxl345-stand_still.csv file
16:37:17  echo: Setting curr from -1 to 800 on stepper_z
16:37:17  // Run Current: 0.80A Hold Current: 0.80A Home Current: 0.20A
16:37:17  echo: Setting curr from -1 to 800 on stepper_z1
16:37:17  // Run Current: 0.80A Hold Current: 0.80A Home Current: 0.20A
16:37:17  echo: Setting curr from -1 to 800 on stepper_z2
16:37:17  // Run Current: 0.80A Hold Current: 0.80A Home Current: 0.20A
16:37:17  echo: Setting curr from -1 to 800 on stepper_z3
16:37:17  // Run Current: 0.80A Hold Current: 0.80A Home Current: 0.20A
16:37:17  echo: Setting tbl from -1 to 2 on stepper_z
16:37:17  echo: Setting tbl from -1 to 2 on stepper_z1
16:37:17  echo: Setting tbl from -1 to 2 on stepper_z2
16:37:18  echo: Setting tbl from -1 to 2 on stepper_z3
16:37:18  echo: Setting toff from -1 to 3 on stepper_z
16:37:18  echo: Setting toff from -1 to 3 on stepper_z1
16:37:18  echo: Setting toff from -1 to 3 on stepper_z2
16:37:19  echo: Setting toff from -1 to 3 on stepper_z3
16:37:19  echo: Setting hend from -1 to 0 on stepper_z
16:37:19  echo: Setting hend from -1 to 0 on stepper_z1
16:37:19  echo: Setting hend from -1 to 0 on stepper_z2
16:37:20  echo: Setting hend from -1 to 0 on stepper_z3
16:37:20  echo: Setting hstrt from -1 to 5 on stepper_z
16:37:20  echo: Setting hstrt from -1 to 5 on stepper_z1
16:37:20  echo: Setting hstrt from -1 to 5 on stepper_z2
16:37:21  echo: Setting hstrt from -1 to 5 on stepper_z3
16:37:21  // CHOPCONF:   33030053 toff=3 hstrt=5 tbl=2 vsense=1 mres=3(32usteps) intpol=1 dedge=1
16:37:21  echo: Speed 5.0 mm/s on 6.54 mm
16:37:21  // accelerometer measurements started
16:37:21  echo: Moving on z axis on 16.535714285714285 mm distance
16:37:25  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_500_41666_1__.csv file
16:37:26  echo: Speed 5.2 mm/s on 6.8 mm
16:37:26  // accelerometer measurements started
16:37:26  echo: Moving on z axis on 16.79714285714286 mm distance
16:37:30  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_520_41666_1__.csv file
16:37:31  echo: Speed 5.4 mm/s on 7.06 mm
16:37:31  // accelerometer measurements started
16:37:31  echo: Moving on z axis on 17.05857142857143 mm distance
16:37:35  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_540_41666_1__.csv file
16:37:35  echo: Speed 5.6 mm/s on 7.32 mm
16:37:35  // accelerometer measurements started
16:37:35  echo: Moving on z axis on 17.32 mm distance
16:37:39  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_560_41666_1__.csv file
16:37:40  echo: Speed 5.8 mm/s on 7.58 mm
16:37:40  // accelerometer measurements started
16:37:40  echo: Moving on z axis on 17.58142857142857 mm distance
16:37:44  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_580_41666_1__.csv file
16:37:45  echo: Speed 6.0 mm/s on 7.84 mm
16:37:45  // accelerometer measurements started
16:37:45  echo: Moving on z axis on 17.84285714285714 mm distance
16:37:48  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_600_41666_1__.csv file
16:37:49  echo: Speed 6.2 mm/s on 8.1 mm
16:37:49  // accelerometer measurements started
16:37:49  echo: Moving on z axis on 18.104285714285716 mm distance
16:37:53  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_620_41666_1__.csv file
16:37:53  echo: Speed 6.4 mm/s on 8.37 mm
16:37:54  // accelerometer measurements started
16:37:54  echo: Moving on z axis on 18.365714285714287 mm distance
16:37:57  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_640_41666_1__.csv file
16:37:58  echo: Speed 6.6 mm/s on 8.63 mm
16:37:58  // accelerometer measurements started
16:37:58  echo: Moving on z axis on 18.627142857142857 mm distance
16:38:01  // Writing raw accelerometer data to /tmp/adxl345-__800_2_3_5_0_-1_660_41666_1__.csv file
16:38:02  !! Move out of range: 159.994 10.000 263.233 [0.000]
16:38:02  !! Move out of range: 159.994 10.000 263.233 [0.000]
16:38:02  !! Move out of range: 159.994 10.000 263.233 [0.000]

MRX8024, 2025-04-06:
> So maybe I just have it configured wrong? On my machine it moves slow, then fast maybe 10 times (with Z set to 260) before I get the out of bounds error. Is there a video of anyone performing this test?

All restrictions are based on the criterion of the time of measurement of vibrations by the accelerometer, in this case the block of code that you provided checks whether all movements fit within the limits of kinematics. It seems that for `find_vibrations` this block is "ignored", since it calculates the movements itself a little higher in the code. 

The test for Z is no different from X or Y, the axis first travels a distance, then returns to the starting position, and so on until it finishes.

yonkiman, 2025-04-06:
Interesting.  What I'm seeing is the test only goes forward (up) - at no point does it go down.  So it hits the limit.

So I assume the slow movements it should be going up, and the faster movements it should be going back down, right?

I'll start looking at that...

MRX8024, 2025-04-06:
> 16:37:58 echo: Moving on z axis on 18.627142857142857 mm distance
> 16:38:02 !! Move out of range: 159.994 10.000 263.233 [0.000]

Try replacing [line](https://github.com/MRX8024/chopper-resonance-tuner/blob/1f98212ca9dbfdf15d516115dd4c26e97b914a8d/chopper_tune.cfg#L387) with
```
G0 {axes[0]}{(minAX + travel_distance)|round(2)} F{speed*60}
```

That's the only thing that can come to my mind

MRX8024, 2025-04-06:
> Interesting. What I'm seeing is the test only goes forward (up) - at no point does it go down. So it hits the limit.

I didn't understand, does your z physically only rise upwards and that's it, or what did you mean?

> So I assume the slow movements it should be going up, and the faster movements it should be going back down, right?

yep

yonkiman, 2025-04-06:
> I didn't understand, does your z physically only rise upwards and that's it, or what did you mean?

That's exactly right.  My z only rises up (slowly, then quickly then slowly then quickly, etc.) until it hits the top of the printer.

yonkiman, 2025-04-06:
Oh shit!  I bet your code assumes G90 and my machine is in G91!

MRX8024, 2025-04-06:
Maybe I misunderstood you in the original message, it doesn't matter anymore. You probably have relative coordinates mode enabled, check.

MRX8024, 2025-04-07:
This is a code oversight, the x-axis move is just a placeholder. In general, where in the gantry you measure vibrations doesn't matter, but it's not like many people have tested and given feedback about the z-axis. You can homing and then manually change the y-position if you want. Btw, it seems like it should go exactly 150 on X.

yonkiman, 2025-04-07:
It does go to 150, once I add the G90 to the start of your macro (operator error again).

Everything's great - already have my first graph.  Thanks so much - I really need this driver tuning, and it's worth it just for the first graph alone (vibration vs speed)!  Tuning the driver will make it even better.  Super helpful.

MRX8024, 2025-04-07:
I hope this script will help you, as it is not a 100% pill, it is just testing different tmc parameters for your steppers. You can share your results here later, if you want.

yonkiman, 2025-04-07:
Definitely helps - thanks.

closed 2025-04-07 by yonkiman