MRX8024/chopper-resonance-tuner #42: FIND_VIBRATIONS option while adjusting stepper currents?

yonkiman, 2025-04-08:
I tried 
`CHOPPER_TUNE FIND_VIBRATIONS=1 CURRENT_MIN_MA=400 CURRENT_MAX_MA=2000 MIN_SPEED=13 MAX_SPEED=13`
hoping it would step through the stepper currents in 25mA increments and plot how the noise varied vs stepper current (this could also be a good way to find the minimum current necessary for a given max speed/accel).  However that command just ran one test at 400mA then stopped testing.  Is there a way to do what I'm trying to do?

Unrelatedly, where are the GBs of data files stored on the host?  Does user have to manually delete them?

MRX8024, 2025-04-08:
The `FIND_VIBRATIONS` command is used only to generate velocity graphs, any other manipulations of registers/currents are performed without it.

The data is stored in `/tmp`, you can see it during testing in the terminal, for example: 
```
Writing raw accelerometer data to /tmp/adxl345-800_2_3_5_0-1_520_41666_1_.csv file
```
At each test start, previous measurements are cleared.

yonkiman, 2025-04-18:
Thanks.  Would be sorta cool to add that feature, but I get that it's outside the scope of this project, which I once again thank you for!  Cheers.

closed 2025-05-19 by yonkiman