MRX8024/chopper-resonance-tuner #45: Measurement procedure

50koruny, 2025-06-20:
First of all, thank you for the great script.
It's a very clever thing.

I have one question – for example, with the Anycubic 4Max Pro printer, I have three different types of motors and all drivers are exactly same TMC2130 from one set.
Basically, the strongest and most stable motor mount is on the Y-axis (attached to the printer body).

What do you think, is it possible to achieve accurate measurements for all axes by simply replacing the motors on the Y axis and measuring them again?
I'm not sure if I explained myself well.
What I mean is to gradually mount all three types of motors on the Y axis and measure them one by one.Without belts being connected.
Then, of course, I would use their specific settings for the axes on which they are actually to be mounted, including the Z-axis and the extruder.

50koruny, 2025-06-20:
Thank you for the quick response.
So when i have 4 pieces of the same drivers, it does not matter which one i will use for the measurements?

MRX8024, 2025-06-20:
Yes. The wiki needs to be improved for a long time, it was meant that there are identical drivers but with different mosfets, sens-resistors, and other nuances that can affect the results. Nothing will change from drivers of the same model and especially batch.